Anonymousdog/displaycameras #95: Simple full-screen alternate between 2 cams?

K4KDR, 2021-02-26:
Thanks for the great utility!  This is going to be a life-saver for me.

I'm embarrassed to say that I have not been able to figure out how to properly setup the **layout.conf.default** file to simply rotate between 2 cameras, showing them full-screen but only one at a time.

The first camera displays properly but it seems no matter what I try, after 5 seconds the second camera just flashes on top of the first.  After another 5 seconds the original image displays properly. 

I know that it's going to seem simple once I understand the syntax how each "Window Position" line should be done, but I've not found the right combination yet.

Thanks much in advance if you could please point me in the right direction on how to set that up.

Best regards,

-Scott

K4KDR, 2021-02-26:
Very sorry - had not seen the **/example_layouts** folder.

One of those addresses my need - thanks again for the great project!!

-Scott

Anonymousdog, 2021-03-05:
No worries. Sorry for the late reply.

Andy

closed 2021-08-18 by Anonymousdog