Anonymousdog/displaycameras #100: picture in picture

ericmersino, 2021-08-01:
I have two feeds. I'm trying to run one full screen, with a second one overlapping that feed on the bottom right corner of the screen. Is there anyway to get this to work and stop flickering / glitching.

SvenVD, 2021-08-01:
This is not about displaycameras, but this can be done with showontop=true in Rpisurv https://gist.github.com/SvenVD/7b973cde2ce5dce6c8284606434e3a14

ericmersino, 2021-08-02:
> This is not about displaycameras, but this can be done with showontop=true in Rpisurv https://gist.github.com/SvenVD/7b973cde2ce5dce6c8284606434e3a14

Thank you, I will check this out and report back

Anonymousdog, 2021-08-02:
No. That would require a custom build with layer support added. I started
to get into that a bit in the layers branch but never completed the work.
If you look at the code, it should be pretty easy to hack together.

Andy

closed 2021-08-18 by Anonymousdog