Anonymousdog/displaycameras #101: Lost camera stream

jpjr1976, 2021-08-02:
First off, Thank you for taking the time to compile this!!!  
Not sure if I'm just missing a step, but everything works great unless a camera goes offline for whatever reason (restart, provision, etc) it will not restart the stream unless i restart the display camera service or reboot the pi.  Is this normal or did i miss something in the setup to "relaunch" a broken stream?

Anonymousdog, 2021-08-03:
Post your layout config file here, and I'll check it. The service should
restart broken feeds automatically. If omxplayer doesn't think it's broken
though, it won't do anything about it.

Thanks,
Andy

jpjr1976, 2021-08-04:
Thanks for taking a look Andy,
Heres what i have, other than the rtsp streams that I removed.  I sent you both files just in case. 

Thanks,
John

# This is the camera feed/windows layout configuration file for the
# displaycameras service.  It ONLY configures the layout and feeds for
# the cameras; the rest of the configuration is in displaycameras.conf.
# See the comments in that file for notes on configuring the below.

# This example defines seven 1/2-HD windows, three of which are off-screen to the right,
# through which the service rotates six camera feeds (it actually uses only six windows)
# on a full-HD monitor.  If this suites your needs, modify only the camera names to taste
# and feed URLs to what your cameras or NVR provides.

# Window names

# 2x2 screen with 3 off-screen windows
windows=(upper_left upper_right lower_left lower_right off_screen off_screen2 off_screen3)
# Make sure to account for each window above in the list below.

# Windows positions

window_positions=(
#First Row
#upper_left
# 680x384 
"0 0 679 383" \
#upper_right
# 680x384
"680 0 1359 383" \

#Second Row (missing all but the far right window because large_left is double size
#lower_left
# 680x384
"0 384 679 767" \
#lower_right
# 680x384
"680 384 1359 767" \

#off-screen
# 680x384 window just off-screen to the right
"1920 0 2879 539" \
# 680x384 window just below the other
"1920 540 2879 1079" \
# 680x384 window just off-screen to the left
"2880 0 3839 539" \
)

# Camera Names

camera_names=(Driveway Garage Driveway Trampoline Backyard)
# Make sure to account for each camera above in the list of feeds below.

# Camera Feeds

camera_feeds=( \
# Mid-Res if your RPi can handle the load
# "rtsp://xxx.xxx.xxx.xxx/yyyyy_1" \
# Low-Res otherwise
# "rtsp://xxx.xxx.xxx.xxx/yyyyy_2" \
#Driveway
"<Camera stream URL>" \
#Garage Driveway
"<Camera stream URL>" \
#Trampoline
"<Camera stream URL>" \
#Backyard
"<Camera stream URL>" \
#West
"<Camera stream URL>" \
#Vestibule
"<Camera stream URL>" \
)

# Are we rotating cameras through the window matrix? (default false if not set here)
# rotate="false"

# This is the configuration file for the displaycameras service intended to
# play a series of camera feeds from Ubiquity camera systems.
# This file sets the global options for the service, but the camera names,
# feeds, display windows, display window names, and rotation are set in the
# display layout files, /etc/displaycameras/layout.conf.default (required)
# and any /etc/displaycameras/layout.conf.<display resolution> you want to support
# when you enable displaydetect in this file.  See the commented layout section
# below for examples to use in the layout files.

#### Global Options

# Do we need screen blanking before displaying cameras (default false)
# This depends on "fbi" being installed.  Blanking fails without it.
#blank="true"

# Optimize reliability by tweaking omx_timeout and/or sleep factor.

# omxplayer network timeout (for establishing feed playback) in seconds.
# I recommend this be no less than 15s and that you increase it as you add
# more than six low-res or four mid-res feeds or add network hops or bandwidth
# challenges to playing the feeds.
omx_timeout=30

# Amount of sleep time we allow per camera when starting|restarting|repairing.
# Recommend no less than 3s start sleep and 1s feedsleep.  You may increase this
# when running higher-res feeds, a large number of cameras, or over slow
# network links.
startsleep=3
feedsleep=1
# Retry is how many seconds of sleep that the script waits after the start or feed
# sleep for omxplayer to first report playback, then playback beyond zero seconds.
retry=5

# Enable display detection (disabled by default) if you setup layout configuration
# files of the naming convention 'layout.conf.<display mode>' (e.g.
# layout.conf.1280x1024).  These files should include window names,
# window positions, camera names, camera feeds, and rotation configuration.
#displaydetect=true

# When rotating camera feeds across a display, how many seconds does each rotation last? (default=5)
#rotatedelay=8

Anonymousdog, 2021-08-04:
> /etc/displaycameras/layout.conf. you want to support when you enable
> configuration files of the naming convention 'layout.conf.' (e.g. layout.conf.1280x1024).

Camera names are the attribute that ties each feed to error recovery. They
can't have spaces. Try "Garage_Driveway" and (just for housekeeping) remove
the feeds that aren't defined and/or aren't used.

Thanks,
Andy

jpjr1976, 2021-08-04:
Andy,
Thank you so much!!!  Worked like a charm!
John

Anonymousdog, 2021-08-04:
You're welcome!

closed 2021-08-18 by Anonymousdog

charlie-greg, 2021-08-22:
I have an interesting dilemma that has plagued me for months. I have a RasPi4 8gig running 4 cameras. The only way I can use the setup is to SSH into the Pi and run HTOP. If I do that it works great. As soon as I stop HTOP after a minute it freezes and eventually crashes. Also, at that point, I loose the SSH connection. BTW, while I'm in HTOP, the cameras run .4%  & the cpu at 2-4%. At this point, I'm looking at purchasing a USB/Ethernet adapter. Another question I have is, can DisplayCameras run on aarch64 (which is what is installed)?

Anonymousdog, 2021-08-22:
Omxplayer isn't supported on 64-bit; so, no.

I'd need a lot more information about your system to diagnose your issue.
The installed service doesn't do what you're reporting. Have you tried any
of the debug instructions?  What OS are you running?

Andy

charlie-greg, 2021-08-23:
Thanks for getting back to me. The Pi 4 is running Buster aarch64 on a Samsung 64G USB drive.
it’s a Model B Rev 1.4
Kernel is 5.10.52-V 8+
Bash 5.0.3
CPU 2.200 GHz
8 Gig of Memory
I’m Running 4 Eufy 2K cameras
On a 1920 x 1080 monitor

I’m curious why it works when I SSH into it and run HTOP.  Even all the clocks are in sync. 
From: Anonymousdog 
Sent: Sunday, August 22, 2021 7:43 PM
To: Anonymousdog/displaycameras 
Cc: charlie-greg ; Comment 
Subject: Re: [Anonymousdog/displaycameras] Lost camera stream (#101)

Omxplayer isn't supported on 64-bit; so, no.

I'd need a lot more information about your system to diagnose your issue.
The installed service doesn't do what you're reporting. Have you tried any
of the debug instructions? What OS are you running?

Andy

Anonymousdog, 2021-08-23:
No idea. I did not create omxplayer, and it's now deprecated. So, if it
doesn't work now, it's not going to work.

Sorry

charlie-greg, 2021-08-23:
Got it. Well, I’m going to create another image that’s is 32 bit and see what happens. As it stands, I know I can get it to work, albeit a not so ideal way. But, I do appreciate your time and input.

Charlie Greg
Canned Jams Radio
412-419-2100  

From: Anonymousdog 
Sent: Monday, August 23, 2021 9:18 AM
To: Anonymousdog/displaycameras 
Cc: charlie-greg ; Comment 
Subject: Re: [Anonymousdog/displaycameras] Lost camera stream (#101)

No idea. I did not create omxplayer, and it's now deprecated. So, if it
doesn't work now, it's not going to work.

Sorry

Anonymousdog, 2021-08-23:
The project is intended for a single-use RPi with no interactive
user...MAYBE doing some other service with very low overhead. That's why
the recommendation for Raspbian-Lite over all other options. If you're
picking a different OS, I have to wonder why.

charlie-greg, 2021-08-24:
I agree. I installed the 32bit Raspian Lite. Same events. In other words, no change.

From: Anonymousdog 
Sent: Monday, August 23, 2021 12:01 PM
To: Anonymousdog/displaycameras 
Cc: charlie-greg ; Comment 
Subject: Re: [Anonymousdog/displaycameras] Lost camera stream (#101)

The project is intended for a single-use RPi with no interactive
user...MAYBE doing some other service with very low overhead. That's why
the recommendation for Raspbian-Lite over all other options. If you're
picking a different OS, I have to wonder why.

Anonymousdog, 2021-08-24:
I don't know. What happens when you try the debug instructions? Does
omxplayer play feeds when launched directly that way?

Andy

charlie-greg, 2021-08-24:
I haven’t seen the debug instructions. Can you point me to them. I will gladly give that a try.

From: Anonymousdog 
Sent: Tuesday, August 24, 2021 9:49 AM
To: Anonymousdog/displaycameras 
Cc: charlie-greg ; Comment 
Subject: Re: [Anonymousdog/displaycameras] Lost camera stream (#101)

I don't know. What happens when you try the debug instructions? Does
omxplayer play feeds when launched directly that way?

Andy

Anonymousdog, 2021-08-24:
At the bottom of the README

charlie-greg, 2021-08-25:
I ran sudo /usr/bin/displaycameras status

and got
cam1 is Playing rtsp://192.168.60.226/live0
at 3393sec
cam2 is Playing rtsp://192.168.60.228/live0
at 64958sec
cam3 is Playing rtsp://192.168.60.230/live0
at 69336sec
cam4 is Playing rtsp://192.168.60.232/live0
at 5014sec

Is this what you mean?

From: Anonymousdog 
Sent: Tuesday, August 24, 2021 1:12 PM
To: Anonymousdog/displaycameras 
Cc: charlie-greg ; Comment 
Subject: Re: [Anonymousdog/displaycameras] Lost camera stream (#101)

At the bottom of the README

Anonymousdog, 2021-08-25:
That's the advanced status information about the running service.

I'm speaking more of the debugging feed problems section of the README. It
details interactive command line testing of omxplayer that you can do from
a ssh session. If your use screen sessions, you can launch all the feeds
concurrently (one in each detached screen session). 'screen' is a Linux app
available in every distro's native repos.

Thanks,
Andy