Anonymousdog/displaycameras #103: any option to blank screen at night and rr-

zeltak, 2021-08-24:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

Anonymousdog, 2021-08-24:
That's outside the scope of this project, but it should be pretty simple to
do as a cron job (not integrated with this system per-se).

Andy

jayjayseal, 2021-12-06:
This is very simple (though beyond this scope of project). I'm using the following setup.

Mind you i'm using this on a pi3. Whenever my homeautomations detect that no one is home i'll let it automatically SSh into the pi and execute the following line: **vcgencmd display_power 0** that disables the HDMI output and let's the screen go into sleep mode. Then do a **vcgencmd display_power 1** when person detection detects someone home again.

This works flawlessly every time and does not reaload the streams as it just disables the HDMI output and re enables it.

zeltak, 2021-12-16:
wow @jayjayseal 

this is exactly what i was looking for

thx so much!

Z

closed 2021-12-16 by zeltak

jayjayseal, 2021-12-16:
Your welcome :-)