Anonymousdog/displaycameras #104: Wont feed on boot

MajorDenkan, 2021-09-15:
I really really think i have done everything right but still do not get ant feed.
I have tested to run OMX player and there i get the feed to work really good with no issues. 

i have attached the layout.conf
[layout.txt](https://github.com/Anonymousdog/displaycameras/files/7167667/layout.txt)

Anonymousdog, 2021-09-15:
You have no feeds configured on the layout file.

MajorDenkan, 2021-09-16:
i can see that i did not get the complete config file.. I have attached it now.
[layout.txt](https://github.com/Anonymousdog/displaycameras/files/7176031/layout.txt)

Anonymousdog, 2021-09-16:
Missing a closing parethesis.

MajorDenkan, 2021-09-17:
Where? As i can see i have a closing parantes after camera feeds...

Anonymousdog, 2021-09-17:
No, you don't.  Put this, ), at the end of the file.

Anonymousdog, 2022-02-16:
omxplayer deprecated in bullseye

closed 2022-02-16 by Anonymousdog