Anonymousdog/displaycameras #105: 14 Camera setup Cameras showing in the past

rickepnet, 2021-10-03:
I have only Ubiquiti using the medium resolution  non-ssl feed. Some cameras show a time stamp of over 20 minutes in the past at times. R PI 4 with 8GB of RAM but only using 32 Raspbian so I only use 4 GB. gpu_mem=512. 

I am not sure it is a bug just wondering why it does that and is there any way to fix it.

Bllstc, 2021-10-13:
Same thing has started happening to me since the last time I updated the RPi a few days ago. Not sure if this is due to displaycameras, omxplayer or if there is something else that causing this as I updated other packages too. But I'm only running displaycameras on that RPi so shouldn't be anything non-raspian lite-standard running on it apart from displaycameras.

I have two Ubiquiti Dome G3 medium feeds on an RPi 2 that have worked flawlessly for several months and they have now started lagging behind just as described above. At first it was only 1/2 feeds that were lagging behind when using the high quality feed, but since I updated the fw on the cameras to 4.30.0 and changed them to medium feed in an effort to resolve this, they are now both lagging behind instead. Behaves like the RPi simply cannot process the feeds in real-time as the framerate is very low on the display, but when checking processes with top command there are plenty of resources available so nothing hogging CPU or memory from what I can see.

stefs22, 2023-02-06:
are someone fix this?

Anonymousdog, 2023-02-07:
No. This project is deprecated.