Anonymousdog/displaycameras #106: Displaycamera does not work after fresh installation

daendaen, 2021-10-09:
Hi,

i installed it on a new SD card, but the service just doesn't work. Did everything as described on your site.

Does anyone have any idea where the error is?

Greetings

`systemctl status displaycameras.service
● displaycameras.service - Displays camera feeds for monitoring
   Loaded: loaded (/etc/systemd/system/displaycameras.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Sat 2021-10-09 13:23:52 CEST; 44s ago
     Docs: man:systemd-sysv-generator(8)
  Process: 587 ExecStart=/usr/bin/displaycameras start (code=exited, status=1/FAILURE)
 Main PID: 587 (code=exited, status=1/FAILURE)

Oct 09 13:23:52 raspberrypi systemd[1]: Starting Displays camera feeds for monitoring...
Oct 09 13:23:52 raspberrypi displaycameras[587]: Please, install omxplayer.
Oct 09 13:23:52 raspberrypi systemd[1]: displaycameras.service: Main process exited, code=exited, status=1/FAILURE
Oct 09 13:23:52 raspberrypi systemd[1]: displaycameras.service: Failed with result 'exit-code'.
Oct 09 13:23:52 raspberrypi systemd[1]: Failed to start Displays camera feeds for monitoring.
`

MajorDenkan, 2021-10-13:
(Oct 09 13:23:52 raspberrypi displaycameras[587]: Please, install omxplayer.)
You need to install OMX player..

z8r-it, 2021-12-18:
omx player is not installable on the newest raspi-os

`Dez 18 11:35:14 raspberrypi displaycameras[6479]: Please, install omxplayer.
Dez 18 11:35:14 raspberrypi systemd[1]: displaycameras.service: Main process exited, code=exited, status=1/>
Dez 18 11:35:14 raspberrypi systemd[1]: displaycameras.service: Failed with result 'exit-code'.
Dez 18 11:35:14 raspberrypi systemd[1]: Failed to start Displays camera feeds for monitoring.
root@raspberrypi:~/displaycameras-master# apt install omxplayer
Paketlisten werden gelesen… Fertig
Abhängigkeitsbaum wird aufgebaut… Fertig
Statusinformationen werden eingelesen… Fertig
Paket omxplayer ist nicht verfügbar, wird aber von einem anderen Paket
referenziert. Das kann heißen, dass das Paket fehlt, dass es abgelöst
wurde oder nur aus einer anderen Quelle verfügbar ist.

E: Für Paket »omxplayer« existiert kein Installationskandidat.
`

LPhil, 2022-01-23:
It looks like the dependent libraries OpenMAX IL and MMAL are deprecated in Bullseye. Is there a plan to fix this? Switching to another player?

https://forums.raspberrypi.com/viewtopic.php?t=324184

Anonymousdog, 2022-01-23:
Not at this time; omxplayer is deprecated. I recommend installing it over
Buster.

Thanks,
Andy

closed 2022-02-16 by Anonymousdog