Anonymousdog/displaycameras #107: displaycameras not working after upgrading to bullseye

vjiggi, 2021-11-20:
**_Before filing a bug report_**

**Replicate your issue on Raspbian Lite**
Raspbian Lite is the preferred target OS for this system.  I will not address bugs that can only be replicated on 'nix with a graphical user interface.

**Ensure your operating system is up-to-date**
...and that you can replicate the problem with an up-to-date (and rebooted) system.
To ensure all is update, run the following in a terminal, `sudo apt-get update && sudo apt-get dist-upgrade -y && sudo apt-get auto-remove -y && sudo apt-get autoclean`

_**If you do the above two things and can still replicate the issue, then...**_

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Install per README.md
2. Configure my feeds in layout.conf.default
3. Run 'sudo displaycameras start'
4. See error "..."

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Raspberry Pi (please complete the following information):**
 - Hardware model: Raspberry Pi 3B+
 - OS: Raspbian_Lite Bullseye
 - Console: ssh
 - gpu memory split: default
 - displaycameras Version [ 0.8.3.3]

**Display Configuration
 - Display/Monitor resolution: [1920x1080]
 - Window matrix: [2x2]
 - Feed resolution: [720 x 480]

**Camera information (please complete the following information):**
 - Number of cameras: [4]
 - Camera models (if not Ubiquiti): Dahua rebranded
 - Managed

**Features enabled (please complete the following information):**
 - Rotation: [no]
 - Display Blanking: [no]
 - Display Detection: [no]

**NVR information (please complete the following information):**
 -  VIP Vision NVR

**Additional context**
stopped working after upgrading from buster to bullseye

**Files
Attach any suspect config files.

vjiggi, 2021-11-20:
It looks like omxplayer is not supported in bullsyeye

Anonymousdog, 2021-11-24:
Yes, this pretty much deprecates this project unless I port it to VLC or
you run Stretch.

vjiggi, 2021-11-26:
Reverted back to buster.

closed 2021-11-26 by vjiggi