Anonymousdog/displaycameras #108: Service saying that its playing stream but nothing showing

PeterPolacek, 2021-11-27:
pi@raspberrypi:~ $ sudo /usr/bin/displaycameras status
NE is Playing http://192.168.1.59:8080/video
at 172sec
SE is NOT playing
South is NOT playing
SW is NOT playing
West is NOT playing
Vestibule is NOT playing

displaycameras.service - Displays camera feeds for monitoring
     Loaded: loaded (/etc/systemd/system/displaycameras.service; enabled; vendor preset: enabled)
     Active: active (exited) since Sat 2021-11-27 16:27:09 GMT; 2min 29s ago
       Docs: man:systemd-sysv-generator(8)
    Process: 4083 ExecStart=/usr/bin/displaycameras start (code=exited, status=0/SUCCESS)
   Main PID: 4083 (code=exited, status=0/SUCCESS)
      Tasks: 10 (limit: 1169)
        CPU: 25.870s
     CGroup: /system.slice/displaycameras.service
             ├─4098 dbus-daemon --fork --print-address 5 --print-pid 6 --session
             ├─4470 /bin/bash /usr/bin/omxplayer --no-keys --no-osd --avdict rtsp_transport:tcp --win 0 0 959 539 http://192.168.1.>
             └─4482 /usr/bin/omxplayer.bin --no-keys --no-osd --avdict rtsp_transport:tcp --win 0 0 959 539 http://192.168.1.59:808>

Nov 27 16:27:05 raspberrypi displaycameras[4449]: Waiting for Vestibule omxplayer startup 1
Nov 27 16:27:06 raspberrypi displaycameras[4449]: Waiting for Vestibule omxplayer startup 2
Nov 27 16:27:07 raspberrypi displaycameras[4449]: Waiting for Vestibule omxplayer startup 3
Nov 27 16:27:08 raspberrypi displaycameras[4449]: Waiting for Vestibule omxplayer startup 4
Nov 27 16:27:09 raspberrypi displaycameras[4449]: Waiting for Vestibule omxplayer startup 5
Nov 27 16:27:09 raspberrypi displaycameras[4449]: Vestibule failed playback
Nov 27 16:27:09 raspberrypi displaycameras[4083]: Camera Display Started
Nov 27 16:27:09 raspberrypi displaycameras[4083]: For complete status info, run
Nov 27 16:27:09 raspberrypi displaycameras[4083]: /usr/bin/displaycameras status.
Nov 27 16:27:09 raspberrypi systemd[1]: Finished Displays camera feeds for monitoring.

But i have no stream on screen.
I am using raspberry pi 3b (16gb sd card, 256mb dedicated gpu memory), external monitor connected via hdmi to raspberry
Raspberry Pi OS Lite
Release date: October 30th 2021
Kernel version: 5.10

Please help!
Thank you

Anonymousdog, 2021-11-27:
Please try the stream debugging steps toward the end of the README.md file.

PeterPolacek, 2021-11-28:
Its seems like omx player wont play stream .. url si fine

Anonymousdog, 2021-11-28:
You may need to change encoding parameters on the camera/DVR end. Omxplayer
only does hardware decode; so, it depends on the codecs licensed in you
particular GPU. They removed a bunch on RPi4. h264 is on all models IIRC.

Figure out the omxplayer command line you'd need, and I'll help you patch
the main script to use the options in the working command.

Thanks,
Andy

closed 2022-02-16 by Anonymousdog