Anonymousdog/displaycameras #109: Flicking Screen

ur7x, 2021-12-01:
**_Before filing a bug report_**

**Replicate your issue on Raspbian Lite**
Raspbian Lite is the preferred target OS for this system.  I will not address bugs that can only be replicated on 'nix with a graphical user interface.

**Ensure your operating system is up-to-date**
...and that you can replicate the problem with an up-to-date (and rebooted) system.
To ensure all is update, run the following in a terminal, `sudo apt-get update && sudo apt-get dist-upgrade -y && sudo apt-get auto-remove -y && sudo apt-get autoclean`

_**If you do the above two things and can still replicate the issue, then...**_

**Describe the bug**
Running Rasbian Lite on a Raspberry Pi4  Displaying 6 800x600 cameras

Seems to lose HDMI sink with the monitor  and the whole thing flickers black running at 1980x1024 @60Hz on a large TV

**To Reproduce**
Steps to reproduce the behavior:
1. Install per README.md
2. Configure my feeds in layout.conf.default
3. Run 'sudo displaycameras start'
4. screen flickers

**Raspberry Pi (please complete the following information):**
 - Pi4
 - OS: Raspbian_Lite 
 - gpu memory split: [e.g. 256MB]
 - displaycameras Version (latest upgrade did not change behaviour)
 - 
**Display Configuration
 - Display/Monitor resolution: [e.g., 1920x1080]
 - Window matrix: [e.g. 1+ 5]
 - Feed resolution: [e.g., low-resolution or 800x600]

**Camera information (please complete the following information):**
 - Number of cameras: [e.g. 6]
 - Camera models : Wifi
 - Unmanaged

**Features enabled (please complete the following information):**
 - Rotation: [no]
 - Display Blanking: [yes]
 - Display Detection: [no]

Anonymousdog, 2021-12-02:
Suggest running just one stream according to debug instructions late in the
README.md. If flickering is present in that debug playback, the problem is
in your display or cabling or RPi.

Thanks

ur7x, 2021-12-02:
Hi thanks for the response.
After about an hour of fiddling with Config.txt settings.. Turning the HDMI_boost on and off (up and down).  I manage to solve the problem by activating the HDMI_1 output and moving the cable there and rebooting.. I think this is either a hardware issue or a bad cable.

KishenPatel1996, 2022-02-25:
For some reason when I add the 5th camera to off screen, my screen just flickers like no other and cameras start overlapping. Any idea why? Sorry for being a noob