Anonymousdog/displaycameras #110: Error while loading

Therealjosephchrzempiec, 2021-12-31:
Hello. I installed Everything the way you said to do in your discription But I'm not getting nothing. So I have tried your testing sudo omxplayer --no-keys --no-osd --avdict rtsp_transport:tcp <camera feed URL> --live -n -1 --timeout 30 with my RTSP and I get this error.

/usr/bin/omxplayer.bin: error while loading shared libraries: libbrcmEGL.so: cannot open shared object file: No such file or directory
I know the RTSP address works because I can see it in VLC player but I have no clue how to fix this problem. I'm not a programmer by any means. Can you please help me?

Edit: I found something online for this error says to update  sudo rpi-update. I did that I get no error now it says Have a Nice day. However I'm still not getting no camera or cameras. I'm now really unsure what to do.

Joseph

Therealjosephchrzempiec, 2021-12-31:
Hello just a update. I'm now getting NE is Playing rtsp://xxx:xxxx@xxx.xxx.xxx.xxx:xxxx at 4sec. But I'm nothing on the screen. I'm trying with one cameras. Here is my config setup.

`# Window names

# 2x2 screen with 3 off-screen windows
windows=(upper_left)
# Make sure to account for each window above in the list below.

# Windows positions

window_positions=(
#First Row
#upper_left
# 960x540
"50 50 959 539" \
)

# Camera Names

camera_names=(NE)
# Make sure to account for each camera above in the list of feeds below.

# Camera Feeds

camera_names=(NE)
# Make sure to account for each camera above in the list of feeds below.

# Camera Feeds

camera_feeds=( \
# Mid-Res if your RPi can handle the load
# "rtsp://xxx.xxx.xxx.xxx/yyyyy_1" \
# Low-Res otherwise
# "rtsp://xxx.xxx.xxx.xxx/yyyyy_2" \
#NE
"rtsp://admin:Pass1230981@10.1.10.35:554" \
)

# Are we rotating cameras through the window matrix? (default false if not set here)
# rotate="true"

# A variable setting how many windows to rotate through in one operation.  Setting this to the number of on-screen windo>
# If you set this higher than the number of cameras, the system just sets it back to 1.  Sane values are 1 through the n>
`

Joseph

Anonymousdog, 2022-01-02:
Did the omxplayer debug (or similar) command ever produce on-screen
playback? No sense in running the service until you get playback during
debug.

Therealjosephchrzempiec, 2022-01-02:
To be honest I have no clue I'm not a programmer. After i started it nothing shows up. I still get the desktop screen and that is all. Status shows it is running but nothing there. Not sure how to do a debug or any commands like that.

Anonymousdog, 2022-01-02:
The debug command right at the top of your first post.

Anonymousdog, 2022-02-16:
This is likely due to running a bullseye distribution or OS that does not support omxplayer.