Anonymousdog/displaycameras #114: displaycameras service not starting

i7phillip, 2022-05-17:
Hello,

I am having an issue getting this working again. I have had it working in the past but when I tried installing on a second pi, I can't get it working. Here is my setup:

Install Raspberry Pi OS Lite (32-BIT Legacy) on SD Card using Raspberry Pi Imager (Raspberry Pi OS – Raspberry Pi) (download the Raspberry Pi Imager to your PC if you have not already done so)

Eject SD Card when writing has completed and reinsert SD Card into reader

Enable WiFi (if needed) Open Notepad or Notepad++
Paste the following (change YOURSSID to the ssid in which you will be connecting to, change YOURPASSWORD [psk] to password of ssid you will be connecting to)
country=US
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1
network={
  ssid="YOURSSID"
  scan_ssid=1
  psk="YOURPASSWORD"
  key_mgmt=WPA-PSK
}
Save as wpa_supplicant.conf in boot partition of SD card

Place SD card into Raspberry Pi and plug in monitor, keyboard and power

Login to Pi using the default username and password
raspberrypi login: pi
Password: raspberry

Change keyboard settings
sudo raspi-config
5 Localisation Options
L3 Keyboard
Genecric 104-key PC>Other>English (US)>press ENTER to select default>No compose key>

Update Raspberry Pi OS
sudo apt-get update && sudo apt-get upgrade -y && sudo reboot

Change password
sudo raspi-config
1 System Options
S3 set new password

Set Time Zone
5 Localisation Options
L2 Timezone
US
Central

Enable SSH
3 Interface Options
P2 SSH
Yes > Ok

Select finish and reboot if prompted

Once rebooted, enter ifconfig at the prompt and note "inet" address under wlan0

Open Command Prompt from your computer and enter
ssh pi@(ip address of pi you wish to connect to)
type "yes" and press enter
enter password for pi

Install OMXPlayer
sudo apt install omxplayer -y

Install displaycameras app by running the following commands
wget https://github.com/Anonymousdog/displaycameras/archive/0.8.3.3.zip
unzip 0.8.3.3.zip
cd displaycameras-0.8.3.3
chmod +x install.sh
sudo ./install.sh
sudo ./install.sh upgrade
When asked about the custom GPU memory split, just press "Enter" to accept the default.
After installation is complete, select y or n to view the README.md.

Open the configuration file using the command prompt
sudo nano /etc/displaycameras/layout.conf.default
Delete the information in the configuration file by selecting all text and press 'ctrl+k'. 
Paste the following into that file:
For main office, copy & paste:
#Main Office Display Setup

#window names

#3x3 Grid

windows=(upper_left upper_middle upper_right left middle right lower_left lower_middle lower_right)

#Window positions

window_positions=(
#First Row
#upper_left
# 640x360
"0 0 639 359" \
#upper_middle
# 640x360
"640 0 1279 359" \
#upper_right
# 640x360
"1280 0 1920 359" \

#Second Row
#left
# 640x360
"0 360 639 719" \
#middle
# 640x360
"640 360 1279 719" \
#right
# 640x360
"1280 360 1920 719" \

#Third Row
#lower_left
# 640x360
"0 720 639 1080" \
#lower_middle
# 640x360
"640 720 1279 1080" \
#lower_right
# 640x360
"1280 720 1920 1080" \
)

camera_names=(CAM01 CAM02 AnotherCam1 CAM07 CAM15 AnotherCam2 CAM12 CAM13 CAM14) 

#These are the camera feeds. Uncomment the cameras you wish to use. Only un-comment the feeds, not the camera name.

camera_feeds=( \
#CAM01
"rtsp://workingurl" \
#CAM02
"rtsp://workingurl" \
#AnotherCam1
"rtsp://workingurl" \
#CAM07
"rtsp://workingurl" \
#CAM15
"rtsp://workingurl" \
#AnotherCam2
#"rtsp://workingurl" \
#CAM12
"rtsp://workingurl" \
#CAM13
"rtsp://workingurl" \
#CAM14
"rtsp://workingurl" \
)

rotate="false"

Save file by pressing 'CTRL+S'.
Exit file by pressing 'CTRL+X'

Anonymousdog, 2022-05-27:
It's very likely you're running RPI Lite Bullseye, and omxplayer isn't
supported on it. You'll need to install RPi Lite Buster to use omxplayer.

Wihanh, 2022-07-29:
is there any plans to migrate the script to use VLC on Bullseye?

Anonymousdog, 2022-07-30:
No. The biggest problem with VLC is the way it registers with DBUS. It will
only register one address which males it impossible to detect and repair
streaming errors.