Anonymousdog/displaycameras #116: Camera not display video

cabli, 2022-06-27:
hello, 
I followed the documentation to install the program but my cameras are not displayed.
when i do the test debug : 
sudo omxplayer --no-keys --no-osd --avdict rtsp_transport:tcp rtsp://192.168.1.1:7447/YZvrRvq39eAmg5qN?enablertp --live -n -1 --timeout 30 
the image is displayed correctly.

But with displaycameras I have :
Starting omxplayer for NE
Waiting for NE omxplayer startup 0
Waiting for NE omxplayer startup 1
Waiting for NE omxplayer startup 2
Waiting for NE omxplayer startup 3
Waiting for NE omxplayer startup 4
Waiting for NE omxplayer startup 5
NE failed playback
and no display video ..
Tanks for help !

[displaycameras.conf.txt](https://github.com/Anonymousdog/displaycameras/files/8995934/displaycameras.conf.txt)
[layout.conf.default.txt](https://github.com/Anonymousdog/displaycameras/files/8995935/layout.conf.default.txt)
o

Anonymousdog, 2022-07-03:
If that's true, then you didn't send me the default layout file at
/etc/displaycameras/ because this one shows more than one camera configured.

Male sure you are editing the correct files.

cabli, 2022-07-03:
hello, yes this is the right file I want to display more than a camera. I tried to comment to show only one camera but the problem remains the same : 

camera_feeds=( \
# Mid-Res if your RPi can handle the load
# "rtsp://xxx.xxx.xxx.xxx/yyyyy_1" \
# Low-Res otherwise
# "rtsp://xxx.xxx.xxx.xxx/yyyyy_2" \
#NE
#"rtsp://192.168.1.1:7447/IZ9UAMZR0yxq0boS?enablertp" \
#SE
"rtsp://192.168.1.1:7447/YZvrRvq39eAmg5qN?enablertp" \
#South
#"rtsp://192.168.1.1:7447/Lq9TlxRLezo8A7x9?enablertp" \
#SW
#"rtsp://192.168.1.1:7447/Lq9TlxRLezo8A7x9?enablertp" \
#West
#"rtsp://192.168.1.1:7447/Lq9TlxRLezo8A7x9?enablertp" \
#Vestibule
#"rtsp://192.168.1.1:7447/Lq9TlxRLezo8A7x9?enablertp" \
)

Anonymousdog, 2022-07-03:
I can't explain how the script would produce failure for the same command
you are issuing to debug (because it's the same command [for omxplayer]).
All this package does is formulate desired RTSP URLs and window sizes
placement.

NicholasBoccio, 2022-07-07:
Have you tried removing the "?enablertp" from each URL? When you change the port as you did, this part of the URL is no longer needed.

MitchJackson94, 2022-07-13:
Looks like your using Unifi protect they have changed it the correct URL is

`rtsp://192.168.1.1:7441/Lq9TlxRLezo8A7x9`

remove s from rtsps change port to 7441 and remove ?enablertp

closed 2022-07-15 by cabli