Anonymousdog/displaycameras #117: Cameras Wont playback

Rogerdoger465, 2022-08-27:
After installation and configuration of the main .conf files, I run the commands to start the streams and it always comes back with an error that omxplayer failed playback on all cameras. We have tested it with the debugging command and it does work with that command. VLC runs the rtsp link perfectly fine, it is purely just the configuration files that run into the issue

the config is not been changed from the already pre-made layout.config.default file, I just added the links and made enabled rotation. The main displaycameras.conf file was not edited since the debugging command worked fine with all 6 streams. I am using a RBPi4 with rasbien buster lite on it. Just ran updates on the OS aswell so there shouldnt be an issue with packages. 

Screenshot of the error i get on all cameras 
 
[image]

Rogerdoger465, 2022-08-27:
The links are as provided, all are the same format 
rtsp://10.232.169.1:7447/VfQT4hnOpxA7IudM

DevNDanger, 2022-09-12:
I am having the same issue. Even after renaming the cameras I am not sure why its pulling up the default cameras.

Anonymousdog, 2022-09-13:
I'll mark up the repo soon enough, but this project is EOL. I'm not
supporting it anymore.

A

DevNDanger, 2022-09-14:
I get it. I wished I new programming better. I just finished my degree but it was mainly java. I find it odd that Ubiquiti doesn't have more support options.

Anonymousdog, 2022-09-14:
Ubiquiti support is IMHO a nightmare as are their many of their upgrades.
It's the whole reason we have only one piece of Ubuquiti equipment in our
firm.

Thanks,
A

MichaelTock, 2022-12-30:
@Rogerdoger465 - not sure if you ever resolved this... I spent a good ~20 mins messing around debugging (it was an RTSPS/7441 vs RTSP/7447 issue) but was still getting the same playback issue. 

In my case, I was updating the `layout.conf.default` in my install folder but it seemed to make no difference. Then I realised it was pulling another default file from `/etc/displaycameras/layout.conf.default`

To resolve this I simply coppied my `layout.conf` file from my original install directory into the `/etc/displaycameras/layout.conf.default` location. Restarted the service and that resolved my issue... up and running. 

FWIW - I am using the older buster OS as per the readme warning.