Anonymousdog/displaycameras #122: [SUGGESTION] whaty about cctv-viewer approach ?

forart, 2023-05-09:
Hi there, 1st of all thanks for your (voluntary) work !

If this project isn't been definitely abandoned, it would be interesting to revive it by "grab inspiration" from @apple-fritter's [RTSP downmix](https://github.com/apple-fritter/RTSP.downmix.4-to-1_2x2) script that renders a 2x2 grid from 4 streams using FFMPEG.

There's also an interesting - but not updated - h264/rtsp [Real Time Video player](https://github.com/eltonwu88/RealTimeVideoPlayer) for Android by @eltonwu88 to evaluate.

Of course there are other interesting RTSP viewers projects such as @iEvgeny's [CCTV Viewer](https://github.com/iEvgeny/cctv-viewer) (that performs extremely smooth on x86 hardware but not on ARM) or @Qrishna's [rtsp_stream_viewer](https://github.com/Qrishna/rtsp-stream-viewer) (which is C++ coded and relays on OpenCV) but are mutch more "articulated"...

...anyway we believe that active collaboration/knowledge exchanges between devs could bring benefits to any project.

Hope that inspires.

apple-fritter, 2023-05-12:
Apologies regarding my lack of understanding the GitHub platform, but maybe this dialogue can continue on in a different venue than this repo's discussion section? I feel though it may be related but is off topic.

forart, 2023-05-12:
> Apologies regarding my lack of understanding the GitHub platform, but maybe this dialogue can continue on in a different venue than this repo's discussion section? I feel though it may be related but is off topic.
>
Well, GH notifies discussion to any "tagged" author (mentioned by "@author")...
...this is one of (working) methods we adopted to stimulate discussion - and hopefully collaboration - between projects' devs/maintainers.

However, since some (you and Evgeny, for example) have reacted positively to our stimulation, we have opened a dedicated discussion here: https://github.com/forart/HyMPS/discussions/17

HussX, 2024-07-02:
It's nowhere as near production ready as displaycamears and it's quite simple, but I came up with a quick replacement for my use cases [here](https://github.com/HussX/piplay).