Anonymousdog/displaycameras #123: losing the rtsp stream

beans4me, 2023-12-21:
I've been using displaycameras for awhile now and it generally runs great but every so often it drops one of the rtsp streams and it doesn't come back unless i restart the service. Is there anyway around this aside scheduling a restart of the device.

Anonymousdog, 2023-12-25:
If omxplayer does not report a failed feed to DBUS, there's nothing to make
the system initiate a feed repair. So, yes. It would be better to sort out
why the feed is dropping though.

beans4me, 2024-11-03:
Can you guide me through troubleshooting? I didn't see your response and just ended up rebooting the pi whenever the feed dropped for the last year or so. Now the feed wont even display on the Pi. I verified the RTSP feed is working by watching it via VLC. The feed in question is from a wireless Unifi G4 doorbell camera. All my other cameras that are hardwired do not have this problem of dropping/not displaying.